ironplc/ironplc #255: Error on check of structured text file with use of enum members

bitkeeper, 2024-06-27:
**Describe the bug**
When using a enum type member it isn't an allowed value.

```ST
TYPE
    MyColors: (Red, Green);
END_TYPE

FUNCTION_BLOCK FB_EXAMPLE
    VAR
        Color: MyColors := Red;
    END_VAR
    Color := Green; (* This isn't allowed; undefined variable *)
END_FUNCTION_BLOCK
```

**To Reproduce**
Steps to reproduce the behavior:

1. Save the above example as test.st:
2. Run:
`ironplc check ./test.st`
3. Output:
```
error[P0015]: Variable not defined before used (variable=Green)
   ┌─ \\?\C:\data\work\staruml\statemachinegenerator\out\test.st:10:14
   │
10 │     Color := Green;
   │              ^^^^^ Undefined variable

Error: "Error during analysis"
```
**Expected behavior**
That the check passes like:
```
OK
```
**System**
 - Version: 0.66.0
 - OS: Windows 11
 - Distribution: Installer

garretfick, 2024-06-27:
Thank you for reporting the issue.

The issue does look genuine given the data provided although I haven't yet tried to reproduce the issue. I will look into the cause and if it is genuine, I'll fix in an upcoming release. Releases normally happen every Monday.

Lastly, a word of caution. IronPLC's semantic analysis is nascent. You are likely to find additional issues. Further reports are very much appreciated.

garretfick, 2024-06-27:
I have reproduced the error. The problem is that by the time semantic analysis gets this far, it thinks that the value is a variable name instead of an enumeration value.

garretfick, 2024-06-27:
The problem is the parser - doesn't implement expression_value as a possible type.

garretfick, 2024-07-01:
Fixing this is turning out to be more complex and I've had less free time than hoped over the weekend. This remains my top item to fix.

garretfick, 2024-07-03:
I have hacked in a fix. The real problem is that I need to build a real symbol table to determine the type of each ambiguous item. That's a way off hence the hacked in fix.

I've kicked off a build so this should be the current release shortly.

closed 2024-07-03 by garretfick

bitkeeper, 2024-07-03:
Thanks for the quick fix!