ironplc/ironplc #275: Question: why create a compiler for your runtime instead of using Rusty?

TheColonel2688, 2024-08-20:
I don't see that discussions are enabled on this repo, so making an issue.

Why create a compiler for your runtime instead of using RuSTy? Rusty biggest hurtle to being useful is that they are not providing a Runtime, or IDE integration because they want to make that closed source. Why not build a runtime and IDE integration for Rusty? Seems that would be more efficient.

Just wondering.

garretfick, 2024-08-20:
An issue works fine.

The reason was I started this project to satisfy some personal learning objectives:
* Learn Rust
* Build a compiler

I didn't seriously look for existing implementations because that wouldn't have satisfied my objectives. That said, even if I knew about RusTy, I wouldn't have used it for a couple reasons:
* license
* no way to update code while running

This is no way criticism of RusTy. I haven't used it because I've been building for my own learning. RusTy may be targeting different use cases.

For this project, would I accept a contribution that adds support for RusTy? Probably yes, but best to discuss first.

TheColonel2688, 2024-08-25:
FYI, there have been recent PRs for allowing Hot-reload/online changes in Rusty.

Sadly the license sucks, but atleast it is only LGPL and not GPL. 

MIT or MPL2 would have been much better choices.

garretfick, 2024-08-25:
Thanks the the info. Hot reload was a key feature.

Agreed on the license.